PRBonn/4DMOS #1: TypeError: cannot pickle 'MinkowskiConvolutionFunction' object

MaxShi007, 2022-06-22:
GREAT WORK！
`
Traceback (most recent call last):
  File "scripts/train.py", line 89, in <module>
    main()
  File "/usr/local/envs/4dmos/lib/python3.8/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/envs/4dmos/lib/python3.8/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/usr/local/envs/4dmos/lib/python3.8/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/envs/4dmos/lib/python3.8/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "scripts/train.py", line 85, in main
    trainer.fit(model, data, ckpt_path=checkpoint)
  File "/usr/local/envs/4dmos/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 770, in fit
    self._call_and_handle_interrupt(
  File "/usr/local/envs/4dmos/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 721, in _call_and_handle_interrupt
    return self.strategy.launcher.launch(trainer_fn, *args, trainer=self, **kwargs)
  File "/usr/local/envs/4dmos/lib/python3.8/site-packages/pytorch_lightning/strategies/launchers/spawn.py", line 78, in launch
    mp.spawn(
  File "/usr/local/envs/4dmos/lib/python3.8/site-packages/torch/multiprocessing/spawn.py", line 230, in spawn
    return start_processes(fn, args, nprocs, join, daemon, start_method='spawn')
  File "/usr/local/envs/4dmos/lib/python3.8/site-packages/torch/multiprocessing/spawn.py", line 179, in start_processes
    process.start()
  File "/usr/local/envs/4dmos/lib/python3.8/multiprocessing/process.py", line 121, in start
    self._popen = self._Popen(self)
  File "/usr/local/envs/4dmos/lib/python3.8/multiprocessing/context.py", line 284, in _Popen
    return Popen(process_obj)
  File "/usr/local/envs/4dmos/lib/python3.8/multiprocessing/popen_spawn_posix.py", line 32, in __init__
    super().__init__(process_obj)
  File "/usr/local/envs/4dmos/lib/python3.8/multiprocessing/popen_fork.py", line 19, in __init__
    self._launch(process_obj)
  File "/usr/local/envs/4dmos/lib/python3.8/multiprocessing/popen_spawn_posix.py", line 47, in _launch
    reduction.dump(process_obj, fp)
  File "/usr/local/envs/4dmos/lib/python3.8/multiprocessing/reduction.py", line 60, in dump
    ForkingPickler(file, protocol).dump(obj)
TypeError: cannot pickle 'MinkowskiConvolutionFunction' object`
This happens when i set N_GPU: 2

benemer, 2022-06-22:
Hey,
Thank you.

Does it work with ```N_GPUS: 1```? I always trained on a single GPU and never encountered this problem.

MaxShi007, 2022-06-22:
> Hey, Thank you.
> 
> Does it work with `N_GPUS: 1`? I always trained on a single GPU and never encountered this problem.

yes,every thing is ok when i use single gpu

benemer, 2022-06-22:
Alright, I see. I always use a single GPU and can not comment on how to make it work with multiple GPUs using the MinkowskiEngine. Therefore, I removed the ```N_GPUS``` features in the latest commit for now and use a single GPU by default.

Feel free to do a PR if you make it work with multiple GPUs. Some resources might be [this](https://nvidia.github.io/MinkowskiEngine/demo/multigpu.html) and [this](https://pytorch-lightning.readthedocs.io/en/1.4.0/advanced/multi_gpu.html).

Thank you!

closed 2022-06-22 by benemer