PRBonn/4DMOS #4: Problem with pretrained model

Gatsby23, 2022-07-05:
Dear Professor:
         Thank you for your nice work about identify the moving objects on the ground. However, when I use the pretrained models, it shows that "it has error in loading state_dict for MOSNet". I don't know why and how to solve this problem. Is there some advice?
       
[image]
 Yours
Qi Wu

benemer, 2022-07-05:
Hey Qi Wu,

Thank you for your interest in our work. The missing key in the checkpoint state dictionary is the loss weight added in https://github.com/PRBonn/4DMOS/pull/3. We just updated the pre-trained models with this parameter. You can now download and use the models as described in the Readme.

Best
Benedikt

Gatsby23, 2022-07-06:
> Thank you for your interest in our work. The missing key in the checkpoint state dictionary is the loss weight added in #3. We just updated the pre-trained models with this parameter. You can now download and use the models as described in the Readme.

Sorry for troubling you again: I use the command "python scripts/predict_confidences.py -w model/5_scans.ckpt
" to do the prediction and the 5_scans.ckpt is the model I download from readme. However, it shows the same error again. The torch and cuda I used are all the same with the Dockfile.

benemer, 2022-07-06:
Did you re-download the model? For me, it does work now. Please verify that you download and use the latest model from our server.

Gatsby23, 2022-07-06:
> Did you re-download the model? For me, it does work now. Please verify that you download and use the latest model from our server.

OK，I'll redonwload it at once. Thank you very much.

benemer, 2022-07-06:
Does it work for you now?

Gatsby23, 2022-07-06:
> Does it work for you now?
[image]
OK, it's work now

Gatsby23, 2022-07-06:
> Does it work for you now?

Thank you very much

closed 2022-07-06 by Gatsby23

MinChoi0129, 2025-01-27:
I had the same error, but I added just one more parameter.
In predict_confidence.py : line 74,

[from]
model.load_state_dict(ckpt["state_dict"])

[to]
model.load_state_dict(ckpt["state_dict"], strict=False)