PRBonn/4DMOS #6: No module named mos4d

E4CEeg, 2022-07-24:
[image]

Hello, i am trying to run 4dmos in google colab. However it shows the "no module named mos4d" and can not run the program. Do you have any idea how to solve this problem?

Thank you.

benemer, 2022-07-25:
Hi @E4CEeg,

did you install the project? To do this in colab, just run 

```
!python3 -m pip install .
```

without the ```--editable``` option. Please note that we recommend the use of docker and our provided docker files to build and run the project.

Best regards
Benedikt

silence-tang, 2022-07-26:
Hello, @E4CEeg,
I used my local Visual Studio Code to run the project and met the same problem as yours, then I solved the problem by moving the `mos4d` directory from `/4DMOS-master/src/` to `/4DMOS-master/scripts/`.  Now I can train the model, and I hope this answer may help you. (However, I dont't know whether this change will trigger other problems...)

E4CEeg, 2022-07-26:
> Hi @E4CEeg,
> 
> did you install the project? To do this in colab, just run
> 
> ```
> !python3 -m pip install .
> ```
> 
> without the `--editable` option. You should also pull the latest commit from the repo since you need an additional `__init__.py` file to import the MinkUNet in colab.
> 
> Please note that we recommend the use of docker and our provided docker files to build and run the project.
> 
> Best regards Benedikt

Hello   @benemer Thank you for your reply. Unfortunately, Google Colab does not support Docker.
After cloning the latest repo, now I can not even install the required installations. 
[image]

benemer, 2022-07-26:
> Hello @benemer Thank you for your reply. Unfortunately, Google Colab does not support Docker. 

Yes, what I meant is that you should use docker instead of google colab if you have a GPU. In general, I can not provide 
support for colab because I am not using it in this project.

> After cloning the latest repo, now I can not even install the required installations. 

Which colab runtime are you using? Since colab outputs
```
Failed to build MinkowskiEngine
```
the problem seems to be the installation of the MinkowskiEngine with cuda support in colab.  Please check https://github.com/NVIDIA/MinkowskiEngine/issues/461 and https://github.com/NVIDIA/MinkowskiEngine/issues/95 for further reference. As far as I can see, there is only the way to install the CPU version of the MinkowsiEngine.

If you do not use the GPU/TPU runtime, the command ```!python3 -m pip install .``` will install the CPU_ONLY build set of the MinkowskiEngine:

[image]

In this case, the project is installed successfully and I can import it. Note that cuda will not be available in this setup.

Again, I recommend not using colab but installing the project with Docker if you have a GPU on your computer.

Best
Benedikt

E4CEeg, 2022-07-26:
Hello Benedikt, thank you for your reply. Because of the gpu I have to use the Google Colab. However, I have installed the Minkowski Engine separately.

closed 2022-07-26 by E4CEeg