PRBonn/4DMOS #8: Cannot find the path

E4CEeg, 2022-07-26:
Hello Benedikt, i am sorry for asking every small details. I am trying to run 

!python3 /content/4DMOS/scripts/predict_confidences.py -w /content/MinkowskiEngine/10_scans.ckpt

and got this error

Traceback (most recent call last):
  File "/content/4DMOS/scripts/predict_confidences.py", line 84, in <module>
    main()
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/content/4DMOS/scripts/predict_confidences.py", line 72, in main
    data.setup()
  File "/usr/local/lib/python3.7/dist-packages/mos4d/datasets/datasets.py", line 47, in setup
    train_set = KittiSequentialDataset(self.cfg, split="train")
  File "/usr/local/lib/python3.7/dist-packages/mos4d/datasets/datasets.py", line 172, in __init__
    path_to_seq = os.path.join(self.root_dir, seqstr)
  File "/usr/lib/python3.7/posixpath.py", line 80, in join
    a = os.fspath(a)
TypeError: expected str, bytes or os.PathLike object, not NoneType

I assume this is from the data path. Actually, I have already exported the path : !export DATA=/content/4DMOS/dataset/sequences
and i also tried to change the self.root_path. However, the error still persists.

benemer, 2022-07-26:
Hi, 

Looks like it can not find the path to the sequence. Can you print the ```self.root_dir``` and ```path_to_seq```  variables from line 172 in ```datasets.py```?

Best,
Benedikt

benemer, 2022-07-26:
Seems like in colab, you should define environment variables like this:

```
%env DATA=/content/4DMOS/dataset/sequences
```

Please try this and let me know if it works.

E4CEeg, 2022-07-27:
Hello Benedikt, thank you very much. It works now.

closed 2022-07-27 by E4CEeg