PRBonn/4DMOS #9: How to visualize the result？

Gatsby23, 2022-07-29:
I have test the code on my own data and get the labels. But I don't know how to visualize the result, could you give me some advice about that ?

benemer, 2022-07-29:
Hey @Gatsby23,

you can use our [semantic-kitti-api](https://github.com/PRBonn/semantic-kitti-api) and follow [this](https://github.com/PRBonn/semantic-kitti-api#lidar-based-moving-object-segmentation-lidar-mos) instruction. Clone the repository in your workspace, install the dependencies and then run the following to visualize for example sequence 8:

```bash
cd semantic-kitti-api
./visualize_mos.py --sequence 8 --dataset /mnt/data/kitti-odometry/dataset --predictions /path/to/4DMOS/predictions/ID/POSES/labels/STRATEGY/
```

It will visualize the 3D point clouds as well as the range images including the predictions:
[image]

Best
Benedikt

Gatsby23, 2022-07-29:
> Hey @Gatsby23,
> 
> you can use our [semantic-kitti-api](https://github.com/PRBonn/semantic-kitti-api) and follow [this](https://github.com/PRBonn/semantic-kitti-api#lidar-based-moving-object-segmentation-lidar-mos) instruction. Clone the repository in your workspace, install the dependencies and then run the following to visualize for example sequence 8:
> 
> ```shell
> cd semantic-kitti-api
> ./visualize_mos.py --sequence 8 --dataset /mnt/data/kitti-odometry/dataset --predictions /path/to/4DMOS/predictions/ID/POSES/labels/STRATEGY/
> ```
> 
> It will visualize the 3D point clouds as well as the range images including the predictions: [image]
> 
> Best Benedikt

Thank you for your replay, I'll give a try.

closed 2022-07-29 by Gatsby23