PRBonn/4DMOS #10: Predicted labels do not appear.

E4CEeg, 2022-08-10:
Hello,
Thank you for your work again. I would like to know how the test data annotated. I have annotated Kitti raw data using https://github.com/jbehley/point_labeler this tool and successfully trained my model.  However, after testing the raw sequences and getting the labels when I open the predicted label in annotation tool, it does not give any prediction.

[image]
Just looks like everything is gray. 

I have tried to change def to_label in confidences_to_label.py. But i have no luck. Can you please look into this ?
Thank you.

benemer, 2022-08-11:
Hi @E4CEeg,

Did you verify that your annotated data is correct? One way to check would be to use the visualizer of the [SemanticKITTI API](https://github.com/PRBonn/semantic-kitti-api). You can also use this tool to visualize your MOS predictions. I never used the point labeler to visualize predictions.

Best
Benedikt

benemer, 2022-08-13:
Did you find any problem on our side? Otherwise, I would close this issue since it seems to be related to the point labeling rather than the 4DMOS code.

E4CEeg, 2022-08-14:
Hello,

As you have suggested, I have tried visualizer of the SemanticKitti API. My annotated labels are shown in the visualizer.
However, my predictions are not shown it. First, I have trained the model with my annotated data. With my trained model, I have tried to predict my own custom data and sequence 11 of SemanticKitti. Both of them do not show any result in the visualizer. Then I have used 10scans.ckpt to predict my own data and same as sequence 11 from SemanticKitti. However, still the results of both of them do not show any moving object.

benemer, 2022-08-14:
Which labels are you using for annotation?

It should definitely work predicting moving objects with the pre-trained ```10scans.ckpt``` model on sequence 11 and visualizing them with the SemanticKITTI API. Can you post the commands you use for running the inference scripts and what your predictions folder looks like?

E4CEeg, 2022-08-17:
Hello, sorry for the late reply. 
I am also confused about the result. 
Here i am running the given trained model.
[image]
[image]
[image]
[image]
[image]

benemer, 2022-08-17:
By running ```visualize.py```, you are using the semantic segmentation config of SemanticKITTI and not the MOS config file. The problem is the wrong label mapping since moving objects are labeled with the class ID 251 which is not present in the SemanticKITTI config for semantic segmentation. Therefore, no labels are displayed. You need to run either

```
./visualize_mos.py --sequence 8 --dataset /path/to/dataset
```

while providing labels in the labels folder or you specify the path to the predictions as described in the [README](https://github.com/PRBonn/4DMOS/edit/master/Readme.md#evaluation-and-visualization):

```
./visualize_mos.py --sequence 8 --dataset /path/to/dataset --predictions /path/to/4DMOS/predictions/ID/POSES/labels/STRATEGY/
```

closed 2022-08-17 by E4CEeg