PRBonn/4DMOS #11: Ground truth poses are not available.

e1339g, 2022-09-06:
Hello, Thank you for your great work and sharing the code.
I have tried to train my own dataset, unfortunately it gives an error "Ground truth poses are not available" and "IndexError: index 0 is out of bounds for axis 0 with size 0". About my dataset, I have downloaded kitti raw data and got my poses file using this script. https://github.com/jbehley/point_labeler/blob/master/scripts/kitti_raw2odometry.py

Here, you can see my folder structure and poses.txt file.
In poses.txt file, each row has 12 elements.
[image]
Thank you.

e1339g, 2022-09-06:
Thank you for your great work, and sharing the code.
I am trying to train my own dataset from kitti raw data. I have converted raw data to semantickitti data format using this script.
https://github.com/jbehley/point_labeler/blob/master/scripts/kitti_raw2odometry.py 

However, when I tried to train the data is gives me an error "Ground truth poses are not available."
and
 inv_frame0 = np.linalg.inv(poses[0])
IndexError: index 0 is out of bounds for axis 0 with size 0

My poses.txt is in same folder in velodyne and label files. And in poses.txt file each row has 12 elements. I do not quite understand why it can not read the poses.txt file. Please help me to train the data
[image]

Thank you

benemer, 2022-09-06:
Hi @e1339g,

This error occurs when the pose file can not be found. Did you follow the instructions in the [Readme](https://github.com/PRBonn/4DMOS#running-the-code) and exported the path to the dataset with

```bash
export DATA=path/to/dataset/sequences
```

If yes, please print the ```pose_path``` that is passed to the ```load_poses``` function to see what is going on.

Best regards
Benedikt

closed 2022-09-06 by e1339g