PRBonn/4DMOS #12: Error building the docker

shayannikoohemat, 2022-10-06:
Thanks for the great work!

When I try building the docker I get the below error which is related to building MinkowskiEngine wheels. I checked on their webpage there are different types of errors related to cuda and pytorch but this error is different. Since I am using the docker in this project I was wondering if anyone encountered similar issue.

```
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.5 LTS
gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0

```
I only copied the error section:

`make build`

```
... 
Building wheel for MinkowskiEngine (setup.py): still running...
  Building wheel for MinkowskiEngine (setup.py): finished with status 'error'
  **ERROR: Command errored out with exit status 1**:
   command: /opt/conda/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-arjsv0pu/minkowskiengine_1b30d99509b54b2eb496604f16599cf7/setup.py'"'"'; __file__='"'"'/tmp/pip-install-arjsv0pu/minkowskiengine_1b30d99509b54b2eb496604f16599cf7/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-dgmhj87k
       cwd: /tmp/pip-install-arjsv0pu/minkowskiengine_1b30d99509b54b2eb496604f16599cf7/
  Complete output (230 lines):
  WARNING: Skipping MinkowskiEngine as it is not installed.
  --------------------------------
  | CUDA compilation set         |
  --------------------------------
  
  Using BLAS=openblas
  Using the default compiler
  running bdist_wheel
  /opt/conda/lib/python3.7/site-packages/torch/utils/cpp_extension.py:381: UserWarning: Attempted to use ninja as the BuildExtension backend but we could not find ninja.. Falling back to using the slow distutils backend.
    warnings.warn(msg.format('we could not find ninja.'))
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-3.7
  creating build/lib.linux-x86_64-3.7/MinkowskiEngine
  copying ./MinkowskiEngine/diagnostics.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine
  copying ./MinkowskiEngine/MinkowskiConvolution.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine
  copying ./MinkowskiEngine/MinkowskiCoordinateManager.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine
  copying ./MinkowskiEngine/MinkowskiBroadcast.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine
  copying ./MinkowskiEngine/MinkowskiUnion.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine
  copying ./MinkowskiEngine/MinkowskiNetwork.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine
  copying ./MinkowskiEngine/MinkowskiKernelGenerator.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine
  copying ./MinkowskiEngine/MinkowskiFunctional.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine
  copying ./MinkowskiEngine/MinkowskiNonlinearity.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine
  copying ./MinkowskiEngine/MinkowskiPooling.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine
  copying ./MinkowskiEngine/MinkowskiChannelwiseConvolution.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine
  copying ./MinkowskiEngine/MinkowskiCommon.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine
  copying ./MinkowskiEngine/sparse_matrix_functions.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine
  copying ./MinkowskiEngine/MinkowskiTensor.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine
  copying ./MinkowskiEngine/MinkowskiSparseTensor.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine
  copying ./MinkowskiEngine/MinkowskiNormalization.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine
  copying ./MinkowskiEngine/MinkowskiTensorField.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine
  copying ./MinkowskiEngine/MinkowskiOps.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine
  copying ./MinkowskiEngine/__init__.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine
  copying ./MinkowskiEngine/MinkowskiInterpolation.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine
  copying ./MinkowskiEngine/MinkowskiPruning.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine
  creating build/lib.linux-x86_64-3.7/MinkowskiEngine/utils
  copying ./MinkowskiEngine/utils/quantization.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine/utils
  copying ./MinkowskiEngine/utils/coords.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine/utils
  copying ./MinkowskiEngine/utils/torchsummary.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine/utils
  copying ./MinkowskiEngine/utils/gradcheck.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine/utils
  copying ./MinkowskiEngine/utils/__init__.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine/utils
  copying ./MinkowskiEngine/utils/init.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine/utils
  copying ./MinkowskiEngine/utils/collation.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine/utils
  copying ./MinkowskiEngine/utils/summary.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine/utils
  creating build/lib.linux-x86_64-3.7/MinkowskiEngine/modules
  copying ./MinkowskiEngine/modules/senet_block.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine/modules
  copying ./MinkowskiEngine/modules/resnet_block.py -> build/lib.linux-x86_64-3.7/MinkowskiEngine/modules
[... 203 lines cut ...]
    Running setup.py install for MinkowskiEngine: started
    Running setup.py install for MinkowskiEngine: still running...
    Running setup.py install for MinkowskiEngine: finished with status 'done'
  DEPRECATION: MinkowskiEngine was installed using the legacy 'setup.py install' method, because a wheel could not be built for it. A possible replacement is to fix the wheel build issue reported above. You can find discussion regarding this at https://github.com/pypa/pip/issues/8368.
  Running setup.py develop for mos4d
Successfully installed Click-8.1.3 MarkupSafe-2.1.1 MinkowskiEngine-0.5.4 PyYAML-6.0 absl-py-1.2.0 aiohttp-3.8.3 aiosignal-1.2.0 async-timeout-4.0.2 asynctest-0.13.0 attrs-22.1.0 cachetools-5.2.0 charset-normalizer-2.1.1 frozenlist-1.3.1 fsspec-2022.8.2 google-auth-2.12.0 google-auth-oauthlib-0.4.6 grpcio-1.49.1 importlib-metadata-5.0.0 markdown-3.4.1 mos4d multidict-6.0.2 ninja-1.10.2.4 oauthlib-3.2.1 packaging-21.3 protobuf-3.19.6 pyDeprecate-0.3.2 pyasn1-0.4.8 pyasn1-modules-0.2.8 pyparsing-3.0.9 pytorch-lightning-1.7.7 requests-oauthlib-1.3.1 rsa-4.9 tensorboard-2.10.1 tensorboard-data-server-0.6.1 tensorboard-plugin-wit-1.8.1 torchmetrics-0.10.0 tqdm-4.64.1 typing-extensions-4.3.0 werkzeug-2.2.2 yarl-1.8.1 zipp-3.8.1
Removing intermediate container 9f66deda49e7
Removing intermediate container e9d102faadec
 ---> bec9b29ccb30
Step 19/20 : RUN adduser --disabled-password --gecos '' --uid $USER_ID --gid $GROUP_ID user
 ---> Running in e4c8f059ced8
Adding user `user' ...
Adding new user `user' (1006) with group `user' ...
Creating home directory `/home/user' ...
Copying files from `/etc/skel' ...
Removing intermediate container e4c8f059ced8
 ---> d55d1ceab527
Step 20/20 : USER user
 ---> Running in 5fb038b1388c
Removing intermediate container 5fb038b1388c
 ---> 090248160fe3

Successfully built 090248160fe3
Successfully tagged mos4d:latest
```

shayannikoohemat, 2022-10-10:
Here is the `make test` results. You are right the installation is finally successful and the docker seems working but I was wondering this error has some further consequences. 
```
$ make test
NVIDIA and CUDA setup
Mon Oct 10 12:22:01 2022       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 470.141.03   Driver Version: 470.141.03   CUDA Version: 11.4     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:0A:00.0 Off |                  N/A |
|  0%   30C    P8    25W / 370W |     19MiB / 24267MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   1  NVIDIA GeForce ...  Off  | 00000000:0B:00.0 Off |                  N/A |
|  0%   29C    P8    19W / 370W |      5MiB / 24268MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
+-----------------------------------------------------------------------------+
Pytorch CUDA setup installed?
True
MinkowskiEngine installed?
/opt/conda/lib/python3.7/site-packages/MinkowskiEngine/__init__.py:42: UserWarning: The environment variable `OMP_NUM_THREADS` not set. MinkowskiEngine will automatically set `OMP_NUM_THREADS=16`. If you want to set `OMP_NUM_THREADS` manually, please export it on the command line before running a python script. e.g. `export OMP_NUM_THREADS=12; python your_program.py`. It is recommended to set it below 24.
  "It is recommended to set it below 24.",
0.5.4
```

benemer, 2022-10-10:
Great!

The GPG key thing is just a warning that it can not pull updates from the Nvidia repositories. But this is fine because they are not updated in the docker container. You can safely ignore it.

Best
Benedikt