PRBonn/4DMOS #15: Trying 4DMOS on already aligned chunks

shayannikoohemat, 2022-10-20:
Hi Benemer,

I have several LAS files which are already registered in a global `UTM-zone35N` coordinate system. To use 4DMOS, I create chunks of my data with 250k points in each chunk **with no overlap** and convert it to bin format. Since the data is aligned, there is no pose information, so I used pretrained model `D=5_scans_no_poses.ckpt` to run the inference. Unfortunately, the results are not good at all. See attached images. I even offset the las files before chunking to the origin=[0, 0, 0] to make sure coordinates are not very large but it didn't help. Do you have any suggestions? How can I create poses for aligned data and use other pretrained models? Or that's not a problem at all?

[image]
                                                                    **Image of three consecutive chunks.**

[image]
                                         **Results of the predictions with no poses. Red points are predicted MOS.**

[image]
                                                                                   **Expected results**

benemer, 2022-10-22:
Hi @shayannikoohemat,

Thanks for your issue. First of all: If the points are already registered, you can pass the `--transform False` flag when running `predict_confidences.py` to not transform the scans again. You should also use another pre-trained model since the one you used expects non-transformed point clouds (and performs worse). Also, you do not need the origin to be `(0,0,0)` as long as there is no overflow if the coordinates are too large. Our model just uses the coordinates for voxelization when generating the sparse tensor, but it is not used as an input to the model.

If I understand you correctly, you have a large point cloud containing multiple registered scans. Next, you cut non-overlapping chunks of this point cloud and  save them as individual point clouds. The problem is that our approach expects sequential data, meaning that the input consists of multiple scans from subsequent timestamps. In your case, the difference between two `.bin` files is not the timestamp but the global location in the original point cloud.

To use 4DMOS, you need to provide sequential data where each file contains the points at a specific timestamp. If you can retrieve timestamps from your data, you could reorganize the files and then pass it to our model.

I hope I could help you.

Best
Benedikt

shayannikoohemat, 2022-10-24:
Thank you Benedikt for your answer. I tried the model on my data with another pretrained model which doesn't need poses using the `--transform False`, it works but still there are poor results (a lot of False Positives). 

Now I'm gonna try your other suggestion and create chunks using timestamps. The current chunks are split by number of points per chunk and since points in a las file are sorted by time so it kind of represents the temporal sequences similar to Kitti structure. In kitti there is no time stamp in bin files so I don't know what is the time window of each bin file and if there is any temporal overlap? Do you have any comment on that?

I can chunk my data let'e say every three meters (using the trajectory) with 4 seconds time window. This gives me chunks with spatial and temporal overlap. I'll try this and let you know. Does it matter for your model if some points are repeated in consecutive files?

benemer, 2022-10-25:
Can you visualize a single chunk of your data? Which sensor was used for capturing the point clouds?

If the points are temporally ordered, your idea of chunking should work if the temporal resolution is chosen such that the movement of an object can be seen in consecutive registered scans. In SemanticKITTI (or KITTI Odometry respectively), the scans come from a rotating scanner and are already motion-compensated, meaning that the points in one `.bin` file approximate the environment at a single timestamp. Therefore, there are no timestamps for each point. The temporal resolution between scans is 0.1 seconds since a LiDAR scanner with 10 Hz framerate was used. This can be different in your case.

If points are repeated in consecutive files, the model will most likely predict these to be static (because they did not move over time, so they should belong to a currently non-moving object).

shayannikoohemat, 2022-10-25:
I use an HDL-32E also from Velodyne. The only difference with Kitti lidar sensor is the number of channels, for kitti is HDL-64E. I can generate chunks every 0.1 seconds but then the chunks are much sparser than Kitti because I'm using a 32 channel sensor. Isn't that a problem?

In my previous experiments points were not duplicated between chunks. So I try to keep it that way.

Here is an example of the timestamps:

[image]

Here is an example comparing the chunk that I generated (right) and Kitti bins:
[image]

shayannikoohemat, 2022-10-27:
Thanks Benedikt for you help. It looks like the problem was the temporal resolution of the sequences. I tried with `0.1` and `0.2` seconds of timestamps to create the splits. The `0.1` timestamps, given our sensor with 32 channels, generates very sparse bins so `0.2` is a better choice (~50k points per bin). I used `10-scans` pretrained model with `non-overlapping` receding strategy. Still some points are remained that can be removed with Knn labeling. Do you have any other suggestion to improve this results?
[image]

benemer, 2022-10-29:
The LiDAR pattern in the chunk you visualized looks like it comes from multiple rotations of the scanner (which will also happen when you put 0.2s of data into  one chunk and your sensor frame rate is 10Hz. This will reduce the MOS performance, since a moving object will appear in the chunk from two different points in time but it will be handled as a single timestamp.

Since you use a LiDAR with 32 beams and the points are ordered in time, you could try to always save chunks of 32x2048 points. This should result in chunks of exactly one rotation of the scanner.

benemer, 2022-10-29:
You can try using our receding horizon strategy and fuse multiple predictions to get a better result.

shayannikoohemat, 2022-10-31:
Many thanks for your feedback.

closed 2022-10-31 by shayannikoohemat