PRBonn/4DMOS #18: Does using closed loop poses have a significant impact on recognition results?

wangfei-tech, 2023-04-01:
Dear author
I used the model you trained 10_ Scans ran on my own computer (not using radar from the Kitti dataset, but their detection accuracy is similar to that of a laser beam) and was pleased that moving objects can be detected, but some moving objects cannot be detected. The positions I used are closed loop poses. Does this closed loop poses I used have an impact on the detection results?
Look forward to your reply！

benemer, 2023-04-02:
Hey,

what do you mean with `not using radar from the Kitti dataset, but their detection accuracy is similar to that of a laser beam`? Are you using radar data?

We experienced some false positives (static objects predicted as moving) when using loop-closed poses. However, this should not affect the false negatives (missed moving objects). But this is hard to say without seeing the data. Can you share the data?

Best
Benedikt

wangfei-tech, 2023-04-02:
Hi, thank you for your timely response！

I used the [MID-360](https://www.livoxtech.com/cn/mid-360) solid-state laser radar in my experiment.
If the experimental results are not good, do I need to re collect data and train the model?

Look forward to your reply! :)

benemer, 2023-04-03:
Got it! Btw, the MID-360 is a LiDAR scanner, not radar ;) That's why I was confused in the beginning.

We previously experienced a similar problem (missed moving objects) with 4DMOS for Livox sensors. We believe this is because of the irregular sensing pattern, which makes it hard to identify a moving object in a limited number of consecutive scans. 

However, we developed a new approach which is currently under review. Since the code is not published yet, you can also send me your data (mersch@igg.uni-bonn.de) and I will have a look if it works better.

closed 2023-04-03 by wangfei-tech