PRBonn/4DMOS #19: Inference problem

lingheng123, 2023-04-03:
when try to use your model to infer,i encountered this error. My pytorchlightning version is 1.9.4.It's the version too high?I saw the requirement is >=1.6.4
[image]

wangfei-tech, 2023-04-03:
I use pytorchlightning version is 1.2.0 it can works corretcly.

benemer, 2023-04-03:
Hey, good catch!

Looks like pytorch_lightning broke backward compatibility at some point since it now expects additional keys when loading a checkpoint from a LightningModule. I personally have version `1.7.7` installed.

I just [pushed](https://github.com/PRBonn/4DMOS/commit/203da9dd5a3ac5db47ab6aba6308305d955ce8dc) the fix to master. We now create an object of the MOSNet class and only load the weights from the checkpoint. Also, I fixed an import that was changed along the way.

I just tested with version `1.9.4`. Let me know if it works for you.

[image]

lingheng123, 2023-04-03:
When I modified it, I solved the error, thank you.

closed 2023-04-03 by benemer