PRBonn/4DMOS #20: evaluate problem

lingheng123, 2023-04-04:
I wanted to evaluate but ran into the following situation. The visualization works normally.
[image]

lingheng123, 2023-04-04:
[image]
[image]

lingheng123, 2023-04-04:
when i use  -s test,i could not find any label
[image]

benemer, 2023-04-04:
What are the sequences you have in your dataset? And how is your split in `onfig/semantic-kitti-mos.yaml`?

The default one is

```yaml
split: # sequence numbers
  train:
    - 0
    - 1
    - 2
    - 3
    - 4
    - 5
    - 6
    - 7
    - 9
    - 10
  valid:
    - 8
  test:
    - 11
    - 12
    - 13
    - 14
    - 15
    - 16
    - 17
    - 18
    - 19
    - 20
    - 21
    ```

so these sequences need to be in both the dataset and your predictions when evaluating.

lingheng123, 2023-04-05:
The data set is set according to the format of the sematic-kitti data set, and the configuration file is unchanged.My prediction sequence is 11 to 21

benemer, 2023-04-05:
Okay. Well in that case, this does not look to be related to 4DMOS. Please check again that you have the predictions and labels specified in the config for the split you select.

For example, in the screenshot of your `sequences` folder, there is not folder for the validation set 08. That's why it can not find the predictions for the validation set.

lingheng123, 2023-04-06:
Yes,when I execute python scripts/confidences_to_labels.py -p predictions/ID/POSES,I choose default sequences (11 to21).That’s why there is no 8 sequence.Can I only use 8 sequence when evaluating IoU?

benemer, 2023-04-06:
Note that you can not evaluate MOS for the test set, because the labels are not public. That's why it can not find them when you run with `-s test`. If you want to get a result here you need to submit your predictions to our benchmark in codalab (see the `semantic-kitti-api` repo for details).

If you want to evaluate the IoU of the validation sequence 08, you can specify it in the `predict_confidences.py` by passing `-seq 8` or change the test split in the config to `08`. Make sure to also select the sequence in the `confidences_to_labels.py` script using the `-seq` flag.

lingheng123, 2023-04-06:
Excuse me, why can't I use pose.txt for 8 sequence to predict?

benemer, 2023-04-06:
Where did I say you can't use the `poses.txt`? Sorry I don't get what you mean

lingheng123, 2023-04-06:
Sorry, did I mean to say that I should use suma pose-txt without loop closure?

benemer, 2023-04-06:
No, this issue is about you having troubles to evaluate the IoU on the predictions. What are you referring to?

In another issue I said that the SuMa poses with loop closing cause false positives on sequence 08 and that you should use the poses without loop closing. But this is another issue and is not related to this one. Please check the other issues and open a new one if you can not find the answer there.

Back to this issue, is it solved now?

lingheng123, 2023-04-06:
yes，I can evalute 8 sequence to get IoU(0.57).I'm sorry to bother you.

benemer, 2023-04-06:
Great! No worries. If you now use the SuMa poses without loop closing for sequence 08 you will get the result reported in the paper. I will close this issue now since it is solved. Just open a new one if another problem arises.

closed 2023-04-06 by benemer