PRBonn/4DMOS #22: Evaluation on 16 channel LiDAR pointcloud

hwan0806, 2023-04-15:
First of all, thank you for great work and code release!

I have one question.
Is there any evaluation result on 16-channel LiDAR pointcloud or have you ever tested on them? 
I tested 4DMOS on my custom dataset(VLP16, OS-64) and the result of OS-64 was awesome. But, in case of VLP-16, this algorithm can't catch moving obejects at all. 
I wonder this caused by my mistake(ex. wrong input poses..) or the lack of various dataset for training!

Thank you!

benemer, 2023-04-17:
Hey, thank you for the feedback :)

Our experiments with sparse LiDAR data (as such of a 16 beam LiDAR) is that MOS is challenging due to the sparse sensor pattern. If the poses are not accurate for the 16 beam scans, this will definitely hurt the MOS performance. Assuming the poses are good enough, a possible reason is that moving objects are not fully covered within the window of past scans.

I never trained on data from other sensor types, because accurate MOS labels are often not available. If you had labeled data from a 16 beam LiDAR, this could definitely help.

We also submitted a new work on MOS which is currently under review. You can send me a data sample if you like and I can see if our new method is capable of segmenting the moving objects with 16 beam scans. My email is mersch@igg.uni-bonn.de.

Best
Benedikt

hwan0806, 2023-04-18:
Thank you for your kind and detailed explanation Benedikt!

I will check the input pose data and LiDAR data again and let you know the result. 
If everything is correct, I will attach my data to the email you told me.
I'm glad to see your great papers every time, and I hope you get a good results for the paper you submitted.

I will close this issue here from now.

benemer, 2023-04-18:
Thank you!

BTW: We estimate the poses inside our pipeline with [KISS-ICP](https://github.com/PRBonn/kiss-icp), so no need to check the poses. You can just send me the data in any common format (.bin, .ply, .pcd, .bag).

closed 2023-04-18 by benemer