PRBonn/4DMOS #24: About the gap between testing result and validation result on SemanticKITTI

liamlin5566, 2023-05-16:
Hi, thanks for sharing your great work.
I am confused with that your validation result is 77.2% shown in paper while your testing result is just 65.2%. Could you explain the gap between testing and validation result? Some issues like #5 indicate that the pose affects the performance significantly. Is it the reason resulting in this gap? Thanks.

liamlin5566, 2023-05-16:
BTW, I noticed that you use KITTI-Road dataset to train and get better testing result. Do you use all KITTI-Road data for training or follow the dataset splitting rule from MotionSeg3D (https://github.com/haomo-ai/MotionSeg3D/blob/master/config/labels/semantic-kitti-mos.yaml).

benemer, 2023-05-17:
> Thanks for replying. I wonder whether you also use the poses without loop closing for training to obtain better result.

No we did not, because we wanted to have exactly the same training setup as other methods.

> BTW, I noticed that you use KITTI-Road dataset to train and get better testing result. Do you use all KITTI-Road data for training or follow the dataset splitting rule from MotionSeg3D (https://github.com/haomo-ai/MotionSeg3D/blob/master/config/labels/semantic-kitti-mos.yaml).

We added all labeled road sequences from [here](https://github.com/haomo-ai/MotionSeg3D/blob/master/config/kitti_road_mos.md) to the training set and obtained better results on the test set. However, we later realized that the road sequence 41 is a subset of test sequence 20. So road sequence 41 should be neither in the training nor in the validation set. I removed this entry from the benchmark. We did not train any other setup so far.

liamlin5566, 2023-05-17:
OK, thanks for answering my questions.

closed 2023-05-17 by liamlin5566