PRBonn/4DMOS #26: Deployment

haider8645, 2023-06-19:
Hello!!

I want to test the model trained on 10 scans in deployment and have a few questions about predict_step function.

1) If I am only interested in the confidence scores of the current scan, would it enough to use logits = out.features_at(0)? Assuming that index 0 points here to 10th scan and index 9 points to the current scan.

2) Do you have some deployment code which I can use to test with new sensor data?
I am testing with the following code. The cuda_tensor_list consists of 10 scans where current scan is at index 9 and 10th scan is at index 0.

            with torch.no_grad():
                out = model.forward(cuda_tensor_list)
                logits = out.features_at(0)
                pred_softmax = F.softmax(logits, dim=1)
                pred_softmax = pred_softmax.detach().cpu().numpy()
                moving_confidence = pred_softmax[:, 2]
                pred_labels = np.ones_like(moving_confidence)
                pred_labels[moving_confidence > 0.5] = 2
                indexes_label_1 = np.where(pred_labels == 1)
                indexes_label_2 = np.where(pred_labels == 2)

                array_of_colors = np.zeros(shape=logits.shape)
                array_of_colors[indexes_label_1] = [0,1,0] # Points with label 1 are colored in Green.
                array_of_colors[indexes_label_2] = [1,0,0] # Points with label 2 are colored in Red.

                print(f"Dynamic Points", {np.sum(pred_labels==2)})
                print(f"Static Points", {np.sum(pred_labels==1)})
                if np.sum(pred_labels == 2) > 0:
                    geom = o3d.geometry.PointCloud()
                    geom.points = o3d.utility.Vector3dVector(points[0][:,:3])
                    geom.colors = o3d.utility.Vector3dVector(array_of_colors)
                    o3d.visualization.draw_geometries([geom])

Best,
Haider

benemer, 2023-06-19:
Hey!

1. `logits = out.features_at(0)` returns the features at batch index 0, not time index. The order of points in the list gets lost, because we stack them into a sparse tensor and only discriminate time using the scan's timestamp. If you only want the predictions for the current timestamp, you can do

```python
mask = coords[:, -1].isclose(current_timestamp)
masked_logits = logits[mask]
```
where `current_timestamp` is the timestamp assigned to the current scan.

2. In your example, `cuda_tensor_list` is expected to be a list of 10 tensors of size [N_i,4], where N_i is number of points of a scan i and 4 are the 4D coordinates x,y,z,t. So you have to provide a timestamp for each scan. In our work, we process data at 10Hz, so 0.1s between scans and we quantize with 0.1 (`self.dt_prediction`) as well. I did not test your code but looks fine from the first view.

You can find an example in our `predict_step` of how to extract predictions for a single timestamp. See [here](https://github.com/PRBonn/4DMOS/blob/a980b40aa33e6b2695432a6aa25ff1b88cee3ace/src/mos4d/models/models.py#L117).

benemer, 2023-06-27:
Can this be closed?

haider8645, 2023-07-04:
Hey, sorry for the late reply. I was on a 2 week vacation. So in the ordering of the 10 samples, the sample at index 0 is the current and the rest of the 9 samples are the last 9 samples?

benemer, 2023-07-17:
I think it is easier to talk about the timestamps instead of the indices. We assign `t=0` to the current scan and all other timestamps are relative to this, so `t=-0.1` for the one before, and so on (assuming `0.1` seconds between two scans).

You can check the `predict_step` [here](https://github.com/PRBonn/4DMOS/blob/a980b40aa33e6b2695432a6aa25ff1b88cee3ace/src/mos4d/models/models.py#L117) and extract the confidence scores at `t=0`, which is the current timestamp.

> Also in context of the 4D data, what are these two functions coordinates_at and features_at actually returning? :)

They return the coordinates and the features at a specific batch index, please check the MinkowskiEngine [documentation](https://nvidia.github.io/MinkowskiEngine/sparse_tensor.html) for details.

Best
Benedikt

haider8645, 2023-07-17:
thanks for reopening the issue. I was able to get some good results from the trained models using the kitti dataset. So now, it would be nice to have a motion tracking system, which tracks the centroids of the detected moving objects. This should help with detecting moving stationary objects.

[image]

[image]

[image]

> I think it is easier to talk about the timestamps instead of the indices. We assign `t=0` to the current scan and all other timestamps are relative to this, so `t=-0.1` for the one before, and so on (assuming `0.1` seconds between two scans).
> 
Ah ok, I understand. If I am using a 5 scans sequence, is it a model requirement that the current scan stamp is t=0? or can the sequence stamps also be e.g. 0.1,0.2,0.3,0.4,0.5 where t=0.5 is the current scan's stamp?

Best,
Haider

benemer, 2023-07-17:
This is also totally fine, the absolute value of timestamps is not important. Just make sure that the time between scans and `DELTA_T_DATA` from the config match.

haider8645, 2023-07-18:
Thank you for the feedback!

closed 2023-07-18 by haider8645