PRBonn/4DMOS #27: Does it have scan_cleaner utils like LM-Net?

Lizhijian-123, 2023-06-19:
Thanks for you code.
Cloud this project get a clean point cloud(no moving object) like Lidar mos?

benemer, 2023-06-19:
Hi!

Yes, you could predict moving objects in the window of scans (with our without the binary Bayes filter) and only add the predicted non-moving points to your map.

Best
Benedikt

benemer, 2023-06-19:
Sorry, I need some more details. What data are you using and what exactly are you trying to do with it?

Lizhijian-123, 2023-06-19:
> Sorry, I need some more details. What data are you using and what exactly are you trying to do with it?

> Sorry, I need some more details. What data are you using and what exactly are you trying to do with it?

Sorry, I didn't express myself to clear it, I am using SemanticKITTI and your Pretrained  Models, and I would like to get a bin file that removes detected moving objects from each point cloud frame.Can this be achieved?

benemer, 2023-06-19:
Okay, that's easy:

1. Use this repository to get the 4DMOS predictions as `.label` files for the desired sequence (instructions in the README).
2. Use [this](https://github.com/PRBonn/LiDAR-MOS/blob/main/utils/scan_cleaner.py) script from LMNet to clean the scans. Note that you need to modify the config file.

closed 2023-06-19 by Lizhijian-123