PRBonn/4DMOS #28: Deploying 4DMOS on Robots

lbhwyy, 2023-07-16:
Hi, 4DMOS is excellent!
I would like to ask how I can deploy 4DMOS on a robot for real-time dynamic object segmentation, so that I can optimize mapping and localization in real-time. Could you provide some methods or suggestions? Thank you!

benemer, 2023-07-17:
Hey, thanks!

This is hard to answer in general. Can you please provide more details on what you want to do? What is the software framework you are working with and which hardware are you using?

lbhwyy, 2023-07-17:
I'm using ROS1 Melodic, and I want to apply 4DMOS on a mobile robot platform. The desired outcome is to be able to dynamically remove objects in real-time using 4DMOS after launching the Velodyne 64-line LiDAR driver, and then perform mapping using Fast LIO.During the localization phase, I can also subscribe to the point cloud topic with dynamic objects removed and then perform NDT (Normal Distribution Transform) algorithm.

benemer, 2023-07-17:
Do you have a GPU available? Otherwise, using the CPU build from the MinkowskiEngine will most likely be too slow for online operation.

In your case, one simple solution would be to wrap 4DMOS into a ROS node that publishes the static point cloud. If you plan to do that, please feel free to open a pull request :)

Best
Benedikt

lbhwyy, 2023-07-21:
Hi,thanks!
I have successfully implemented real-time registration of Velodyne 16-line LiDAR point clouds across multiple frames, and integrated them into the current frame. However, due to limitations in the accuracy and speed of the NDT registration, it is challenging to achieve pose estimation between frames comparable to the KITTI dataset. Therefore, directly applying 4DMOS to Velodyne 16-line LiDAR may not yield satisfactory results, even without considering the differences in data modes between Velodyne 16 and Velodyne 64.
Best
LBH

benemer, 2023-07-23:
You should definitely try our odometry pipeline [KISS-ICP](https://github.com/PRBonn/kiss-icp) which is also released as a ROS 1 package. You can find some discussions on how to modify it for very sparse sensors. It usually gives reasonable results without a lot of parameter tuning and I used it in our latest MOS work to retrieve the pose information.

You can also send me some recorded data and I will give it a try if you like.

Best,
Benedikt

lbhwyy, 2023-07-23:
Thank you! I will try kiss-icp.
Best
LBH

closed 2023-07-31 by benemer