PRBonn/4DMOS #29: Results regarding Apollo

JulesSanchez, 2023-07-31:
Thanks you very much for your work,

In you paper in Table II, you compare your methods with other on the Apollo dataset. Where did you find this dataset and its annotations ? I'd be interested in comparing my methods with yours on your dataset.

Best,

Jules

benemer, 2023-07-31:
Hey Jules,

Please find the data [here](https://www.ipb.uni-bonn.de/html/projects/apollo_dataset/LiDAR-MOS.zip).

The data was converted to KITTI format and labeled by my colleague @Chen-Xieyuanli for our [AutoMOS](https://arxiv.org/pdf/2201.04501.pdf) paper.

By the way, we just published another MOS approach which is called [MapMOS](https://www.ipb.uni-bonn.de/wp-content/papercite-data/pdf/mersch2023ral.pdf). The code will be open-sourced soon and you will be able to additionally compare your method on Nuscenes as well as a very busy KITTI tracking sequence that we labeled.

Stay tuned :)

JulesSanchez, 2023-07-31:
It's perfect and exactly what I needed !

closed 2023-07-31 by JulesSanchez