PRBonn/4DMOS #31: 4D

ChiyunNoh, 2023-09-13:


ChiyunNoh, 2023-09-13:
How can I use 4DMOS in Nuscenes?
Should I have to edit Nuscenes dataset into KITTI format?

benemer, 2023-09-19:
Hey!

You would need to convert it to the KITTI format to use the full 4DMOS training pipeline of this repository. However, for just inferring moving objects it should be straightforward to only take the `MOSModel` from the [`models.py`](https://github.com/PRBonn/4DMOS/blob/master/src/mos4d/models/models.py). Just load a buffer of point clouds with a timestamp attached to each cloud, register them in a common frame and pass the 4D tensor as a batch to the `forward` function. You can check https://github.com/PRBonn/4DMOS/issues/26 and https://github.com/PRBonn/4DMOS/issues/30 for a more detailed discussion. 

You can also check our recent work (`MapMOS`)[https://github.com/PRBonn/MapMOS] for an out-of-the-box moving object segmentation. It is easy to use and works with nuScenes without any modification of code or data conversion.

Best
Benedikt

closed 2023-10-16 by ChiyunNoh