PRBonn/4DMOS #35: Training and inference problems

YuancXie, 2024-01-25:
I tried to train a model using the default setting. However, I got a long list of errors. I suspect that it is data related. I did follow the diagram to prepare the data. In Folders 00 through 10, I have "labels" and "velodyne" sub-folders and "calib.txt" and "poses.txt". I noticed that the "poses.txt" file has the same number of rows as the number of files in "labels" and "velodyne" sub-folders.

For Folders 11 through 21, I only have "velodyne" sub-folder, "calib.txt" and "poses.txt". In this folders, the "poses.txt" file has much more rows than the number of files in the "velodyne" sub-folder. 

I also got errors when running the sample inference command. Once this problem is solved, the other one probably will be gone as well. Could you please help me out? Thank you very much.

(base) :~/4DMOS$ python scripts/train.py                                                               
/home/yc/miniconda3/lib/python3.11/site-packages/MinkowskiEngine-0.5.4-py3.11-linux-x86_64.egg/MinkowskiEngine/__init__.py:36: UserWarning: The environment variable `OMP_NUM_THREADS` not set. MinkowskiEngine will automatically set `OMP_NUM_THREADS=16`. If you want to set `OMP_NUM_THREADS` manually, please export it on the command line before running a python script. e.g. `export OMP_NUM_THREADS=12; python your_program.py`. It is recommended to set it below 24.
  warnings.warn(
GPU available: True (cuda), used: True
TPU available: False, using: 0 TPU cores
IPU available: False, using: 0 IPUs
HPU available: False, using: 0 HPUs
You are using a CUDA device ('NVIDIA GeForce RTX 3080') that has Tensor Cores. To properly utilize them, you should set `torch.set_float32_matmul_precision('medium' | 'high')` which will trade-off precision for performance. For more details, read https://pytorch.org/docs/stable/generated/torch.set_float32_matmul_precision.html#torch.set_float32_matmul_precision
Loaded 19040 training, 4062 validation and 3802 test samples.
LOCAL_RANK: 0 - CUDA_VISIBLE_DEVICES: [0]

  | Name                  | Type                  | Params
----------------------------------------------------------------
0 | model                 | MOSModel              | 1.8 M 
1 | MOSLoss               | MOSLoss               | 0     
2 | ClassificationMetrics | ClassificationMetrics | 0     
----------------------------------------------------------------
1.8 M     Trainable params
0         Non-trainable params
1.8 M     Total params
7.388     Total estimated model params size (MB)
Sanity Checking DataLoader 0:   0%|                                                     | 0/2 [00:00<?, ?it/s]Traceback (most recent call last):
  File "/home/yc/4DMOS/scripts/train.py", line 87, in <module>
    main()
  File "/home/yc/miniconda3/lib/python3.11/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yc/miniconda3/lib/python3.11/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/home/yc/miniconda3/lib/python3.11/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yc/miniconda3/lib/python3.11/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yc/4DMOS/scripts/train.py", line 83, in main
    trainer.fit(model, data, ckpt_path=checkpoint)
  File "/home/yc/miniconda3/lib/python3.11/site-packages/pytorch_lightning/trainer/trainer.py", line 544, in fit
    call._call_and_handle_interrupt(
  File "/home/yc/miniconda3/lib/python3.11/site-packages/pytorch_lightning/trainer/call.py", line 44, in _call_and_handle_interrupt
    return trainer_fn(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yc/miniconda3/lib/python3.11/site-packages/pytorch_lightning/trainer/trainer.py", line 580, in _fit_impl
    self._run(model, ckpt_path=ckpt_path)
  File "/home/yc/miniconda3/lib/python3.11/site-packages/pytorch_lightning/trainer/trainer.py", line 989, in _run
    results = self._run_stage()
              ^^^^^^^^^^^^^^^^^
  File "/home/yc/miniconda3/lib/python3.11/site-packages/pytorch_lightning/trainer/trainer.py", line 1033, in _run_stage
    self._run_sanity_check()
  File "/home/yc/miniconda3/lib/python3.11/site-packages/pytorch_lightning/trainer/trainer.py", line 1062, in _run_sanity_check
    val_loop.run()
  File "/home/yc/miniconda3/lib/python3.11/site-packages/pytorch_lightning/loops/utilities.py", line 182, in _decorator
    return loop_run(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yc/miniconda3/lib/python3.11/site-packages/pytorch_lightning/loops/evaluation_loop.py", line 134, in run
    self._evaluation_step(batch, batch_idx, dataloader_idx, dataloader_iter)
  File "/home/yc/miniconda3/lib/python3.11/site-packages/pytorch_lightning/loops/evaluation_loop.py", line 391, in _evaluation_step
    output = call._call_strategy_hook(trainer, hook_name, *step_args)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yc/miniconda3/lib/python3.11/site-packages/pytorch_lightning/trainer/call.py", line 309, in _call_strategy_hook
    output = fn(*args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^
  File "/home/yc/miniconda3/lib/python3.11/site-packages/pytorch_lightning/strategies/strategy.py", line 403, in validation_step
    return self.lightning_module.validation_step(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yc/4DMOS/src/mos4d/models/models.py", line 98, in validation_step
    loss = self.getLoss(out, past_labels)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yc/4DMOS/src/mos4d/models/models.py", line 54, in getLoss
    loss = self.MOSLoss.compute_loss(out, past_labels)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yc/4DMOS/src/mos4d/models/loss.py", line 35, in compute_loss
    loss = self.loss(log_softmax, gt_labels.long())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yc/miniconda3/lib/python3.11/site-packages/torch/nn/modules/module.py", line 1518, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yc/miniconda3/lib/python3.11/site-packages/torch/nn/modules/module.py", line 1527, in _call_impl
    return forward_call(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yc/miniconda3/lib/python3.11/site-packages/torch/nn/modules/loss.py", line 216, in forward
    return F.nll_loss(input, target, weight=self.weight, ignore_index=self.ignore_index, reduction=self.reduction)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yc/miniconda3/lib/python3.11/site-packages/torch/nn/functional.py", line 2729, in nll_loss
    return torch._C._nn.nll_loss_nd(input, target, weight, _Reduction.get_enum(reduction), ignore_index)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: Expected input batch_size (327680) to match target batch_size (2463436).

benemer, 2024-02-05:
The dataloader can not find your data. How is your data organized, how does your config look like and did you specify the `DATA` variable in your environment?

YuancXie, 2024-02-05:
Thank you for your prompt response. It was a silly mistake I made.
I did "DATA=/home/yc/4DMOS/dataset/sequences" and forgot the "export" at the beginning. That did not trigger any errors/warnings. 
Now the problem is solved. Thank you again.