PRBonn/4DMOS #40: mos4d_pipeline：Command not found

iwander-all, 2024-09-13:
Hi, after install 4DMOS with `make install`, when I run `mos4d_pipeline --help`, it shows `mos4d_pipeline：Command not found`, how to solve it, thanks

iwander-all, 2024-09-13:
install is success:

`Using pip 23.2.1 from /usr/local/anaconda3/lib/python3.11/site-packages/pip (python 3.11)
Defaulting to user installation because normal site-packages is not writeable
Processing /media/hunkyu/SuperBigDisk/DeepLearning/net/MOS/4DMOS
  Running command pip subprocess to install build dependencies
  Collecting setuptools>=61.0
    Obtaining dependency information for setuptools>=61.0 from https://files.pythonhosted.org/packages/cb/9c/9ad11ac06b97e55ada655f8a6bea9d1d3f06e120b178cd578d80e558191d/setuptools-74.1.2-py3-none-any.whl.metadata
    Using cached setuptools-74.1.2-py3-none-any.whl.metadata (6.7 kB)
  Using cached setuptools-74.1.2-py3-none-any.whl (1.3 MB)
  Installing collected packages: setuptools
  ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
  conda-repo-cli 1.0.75 requires requests_mock, which is not installed.
  spyder 5.4.3 requires pyqt5<5.16, which is not installed.
  spyder 5.4.3 requires pyqtwebengine<5.16, which is not installed.
  conda-repo-cli 1.0.75 requires clyent==1.2.1, but you have clyent 1.2.2 which is incompatible.
  conda-repo-cli 1.0.75 requires PyYAML==6.0.1, but you have pyyaml 6.0 which is incompatible.
  Successfully installed setuptools-74.1.2
  Installing build dependencies ... done
  Running command Getting requirements to build wheel
  running egg_info
  writing src/mos4d.egg-info/PKG-INFO
  writing dependency_links to src/mos4d.egg-info/dependency_links.txt
  writing entry points to src/mos4d.egg-info/entry_points.txt
  writing requirements to src/mos4d.egg-info/requires.txt
  writing top-level names to src/mos4d.egg-info/top_level.txt
  reading manifest file 'src/mos4d.egg-info/SOURCES.txt'
  adding license file 'LICENSE'
  writing manifest file 'src/mos4d.egg-info/SOURCES.txt'
  Getting requirements to build wheel ... done
  Running command Preparing metadata (pyproject.toml)
  running dist_info
  creating /tmp/pip-modern-metadata-2ndoipya/mos4d.egg-info
  writing /tmp/pip-modern-metadata-2ndoipya/mos4d.egg-info/PKG-INFO
  writing dependency_links to /tmp/pip-modern-metadata-2ndoipya/mos4d.egg-info/dependency_links.txt
  writing entry points to /tmp/pip-modern-metadata-2ndoipya/mos4d.egg-info/entry_points.txt
  writing requirements to /tmp/pip-modern-metadata-2ndoipya/mos4d.egg-info/requires.txt
  writing top-level names to /tmp/pip-modern-metadata-2ndoipya/mos4d.egg-info/top_level.txt
  writing manifest file '/tmp/pip-modern-metadata-2ndoipya/mos4d.egg-info/SOURCES.txt'
  reading manifest file '/tmp/pip-modern-metadata-2ndoipya/mos4d.egg-info/SOURCES.txt'
  adding license file 'LICENSE'
  writing manifest file '/tmp/pip-modern-metadata-2ndoipya/mos4d.egg-info/SOURCES.txt'
  creating '/tmp/pip-modern-metadata-2ndoipya/mos4d-1.0.0.dist-info'
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: kiss-icp>=1.0.0 in /home/hunkyu/.local/lib/python3.11/site-packages (from mos4d==1.0.0) (1.0.0)
Requirement already satisfied: diskcache>=5.3.0 in /home/hunkyu/.local/lib/python3.11/site-packages (from mos4d==1.0.0) (5.6.3)
Requirement already satisfied: pytorch-lightning>=1.6.4 in /home/hunkyu/.local/lib/python3.11/site-packages (from mos4d==1.0.0) (2.4.0)
Requirement already satisfied: natsort in /home/hunkyu/.local/lib/python3.11/site-packages (from kiss-icp>=1.0.0->mos4d==1.0.0) (8.4.0)
Requirement already satisfied: numpy in /usr/local/anaconda3/lib/python3.11/site-packages (from kiss-icp>=1.0.0->mos4d==1.0.0) (1.24.3)
Requirement already satisfied: plyfile in /home/hunkyu/.local/lib/python3.11/site-packages (from kiss-icp>=1.0.0->mos4d==1.0.0) (1.1)
Requirement already satisfied: pydantic>=2 in /home/hunkyu/.local/lib/python3.11/site-packages (from kiss-icp>=1.0.0->mos4d==1.0.0) (2.9.1)
Requirement already satisfied: pydantic-settings in /home/hunkyu/.local/lib/python3.11/site-packages (from kiss-icp>=1.0.0->mos4d==1.0.0) (2.5.2)
Requirement already satisfied: pyquaternion in /home/hunkyu/.local/lib/python3.11/site-packages (from kiss-icp>=1.0.0->mos4d==1.0.0) (0.9.9)
Requirement already satisfied: rich in /home/hunkyu/.local/lib/python3.11/site-packages (from kiss-icp>=1.0.0->mos4d==1.0.0) (13.8.1)
Requirement already satisfied: tqdm in /usr/local/anaconda3/lib/python3.11/site-packages (from kiss-icp>=1.0.0->mos4d==1.0.0) (4.65.0)
Requirement already satisfied: typer[all]>=0.6.0 in /home/hunkyu/.local/lib/python3.11/site-packages (from kiss-icp>=1.0.0->mos4d==1.0.0) (0.12.5)
Requirement already satisfied: torch>=2.1.0 in /home/hunkyu/.local/lib/python3.11/site-packages (from pytorch-lightning>=1.6.4->mos4d==1.0.0) (2.4.1)
Requirement already satisfied: PyYAML>=5.4 in /usr/local/anaconda3/lib/python3.11/site-packages (from pytorch-lightning>=1.6.4->mos4d==1.0.0) (6.0)
Requirement already satisfied: fsspec[http]>=2022.5.0 in /usr/local/anaconda3/lib/python3.11/site-packages (from pytorch-lightning>=1.6.4->mos4d==1.0.0) (2023.4.0)
Requirement already satisfied: torchmetrics>=0.7.0 in /home/hunkyu/.local/lib/python3.11/site-packages (from pytorch-lightning>=1.6.4->mos4d==1.0.0) (1.4.1)
Requirement already satisfied: packaging>=20.0 in /usr/local/anaconda3/lib/python3.11/site-packages (from pytorch-lightning>=1.6.4->mos4d==1.0.0) (23.1)
[... cut ...]
  Attempting uninstall: mos4d
    Found existing installation: mos4d 1.0.0
    Uninstalling mos4d-1.0.0:
      Removing file or directory /home/hunkyu/.local/bin/4dmos_pipeline
      Removing file or directory /home/hunkyu/.local/lib/python3.11/site-packages/mos4d-1.0.0.dist-info/
      Removing file or directory /home/hunkyu/.local/lib/python3.11/site-packages/mos4d/
      Successfully uninstalled mos4d-1.0.0
  changing mode of /home/hunkyu/.local/bin/4dmos_pipeline to 775
Successfully installed mos4d-1.0.0
`

benemer, 2024-09-13:
Hi,

Sorry, it's `4dmos_pipeline`. I will change it in the README. Thanks!

closed 2024-09-13 by benemer