PRBonn/4DMOS #41: suggestion for install

iwander-all, 2024-09-16:
Hello, 

The install is too hard and I spand days to install and it even disturb my other works' env. I think following is better:

```
conda create -n mos4d python=3.8 -y 
conda install pytorch==1.12.1 torchvision==0.13.1 torchaudio==0.12.1 cudatoolkit=11.6 -c pytorch -c conda-forge
git clone https://github.com/NVIDIA/MinkowskiEngine.git && cd MinkowskiEngine
python setup.py install --blas_include_dirs=${CONDA_PREFIX}/include --blas=openblas **--force_cuda**
```

before make install, "pytorch_lightning>=1.6.4" pyproject.toml should change to "pytorch_lightning==1.6.4", its version should not >2!

benemer, 2024-09-16:
Well, the installation of MinkowskiEngine as a dependency is tricky, not the installation of 4DMOS. Your instructions will not help if someone does not want to use conda. So, I will leave it to the user to figure out how they install the MinkowskiEngine.

Why should the `pytorch_lightning` version be `<2`? Also, if that's what you want, you should not pin it down to a specific version but specify `<2`. But I do not see a problem in using `>=2`.

benemer, 2024-09-19:
I will close this issue. If you still insist on downgrading `pytorch_lightning`, please feel free to re-open :)

closed 2024-09-19 by benemer

iwander-all, 2024-09-19:
Thanks for your help,  I successfully install and run the demo.