PRBonn/4DMOS #43: Scan and Label don't contain same number of points?

pengboyemo, 2024-12-25:
Hi, @benemer 
Below the first picture is the result labels of 4dmos, when i use semantic-kitti-api script visualize_mos.py but get a error like the second picture, why the reason for this?
[image]

[image]

benemer, 2024-12-26:
Which configuration do you use? For evaluation on SemanticKITTI, you can pass the kitti config to classify moving/non-moving for all points in the scan.

As you can see from the error message, there are less labels in your predictions than points Most likely, your max_range for mos was not set to -1 which disables the clipping and/or the min_range was not set to 0.

pengboyemo, 2024-12-26:
Thanks for your replay, below the first file the SemanticKITTI configure file which i changed it's format yaml to txt for uploading, the second file is 4dmos configure file, but i am not sure 4DMOS configure like kitti.yaml or other file is need to be modified? 
[semantic-kitti-mos.txt](https://github.com/user-attachments/files/18251864/semantic-kitti-mos.txt)
[kitti.txt](https://github.com/user-attachments/files/18251876/kitti.txt)

benemer, 2024-12-26:
You do not need to modify the file, just pass the path to the kitti.txt (the 4DMOS config file) using the -c flag when running the pipeline.

pengboyemo, 2024-12-26:
I have get the right result in the below picture, i will close my issue, thanks for your help and appreciate your work.
[image]
[image]

closed 2024-12-26 by pengboyemo