PRBonn/4DMOS #46: labels file does not contain static/dynamic information

curiousityy, 2025-02-24:
Hi @benemer,

I am trying to run 4DMOS on my own dataset and modified the code accordingly. I am generating the .label file (I get 2 folders, 4dmos_online and 4dmos_receding). Both the folders have label files containing info 0-9 and so, when I use this label file with the scan_cleaner from LM-Net, I dont see the segmented dynamic objects (LM-Net filters with a threshold of 250). May I know how can I get the .label file to mimic LM-Net, or, how do I use this .label file to segment dynamic objects?

curiousityy, 2025-02-24:
@benemer, thank you for the quick update. Yes, I also tried with the semantic_kitti_api, but that also doesn't work since the label for moving car in the tool as well is 252 (which is where LM-Net probably get s its threshold from), and so the visualizing there is just the raw scans

benemer, 2025-02-24:
The label should not be a problem. When we save the .bin files, we save it with label 251, which is also considered to be moving by the semantic-kitti API.

Can you check again if there is any output on the terminal when running the visualizer?

curiousityy, 2025-02-24:
Are you looking for any specific output? I just see the visualizer tool fetching bin and labels file. 
Also, you mentioned that you save .bin files with a label 251, but, I don't see .bin files saved anywhere in the results folders. I see only label files. Is there some flag to save .bin files that I am missing?

benemer, 2025-02-24:
I wanted to see if the number of labels and points matches or if the visualizer throws any error/warning.

Sorry, I meant .label files. If you see those files and you can visualize them, the problem is most likely not on the 4DMOS side.

curiousityy, 2025-02-24:
@benemer, yes. The idea is to extract only the static objects from a scene and process that further in my pipeline. I started back tracing the issue since the scan_cleaner should return static pointclouds, but I noticed that the cleaned scans had dynamic objects too. So when I extracted information from the labels file, every label, for every every point, in every scan was 9 (which is static class in KITTI). I did not find a 251/252. The odometry output I get is correct and so I assume every scan is processed, but somehow, every point in all the scans is detected as static.

benemer, 2025-02-24:
Can you check if 4DMOS correctly estimates the moving objects by running the visualizer and also dumping to ply?

curiousityy, 2025-02-24:
@benemer, I was unfortunately not able to run the --visualize flag. I keep getting a "RuntimeError: [polyscope]  [EXCEPTION] Automatic initialization: no Polyscope backends could be initialized successfully". That also one of the reasons why I am trying to visualize with the labels file

benemer, 2025-02-25:
In your first code you said you `modified the code accordingly`, what exactly did you do?

curiousityy, 2025-02-25:
@benemer , Sure. I can share the dataset with you. I will share a small sample where there are definite moving vehicles. Hope that is ok? The dataset in itself is quite huge to share as a whole. 

For the changes, I only modified the way bin files are fetched since the 4dmos repo expects a kitti like folder structure, but my changes were strictly in data fetching. Nothing to do with the model/saving of labels.

benemer, 2025-02-25:
Yes, that's totally fine. Thanks!

benemer, 2025-02-25:
Thanks for sharing the data.

First of all, I encourage you to use the `10_scans. ckpt` model because this achieves the best results according to our ablation study in the paper. I tried with both, and 4DMOS struggled to find the moving objects. This is I believe because of the sparsity of the data.

But it does also work for some cases, see here for the moving pedestrians:

[image]

I suggest you to use our newer Map-based moving object segmentation pipeline called [MapMOS](https://github.com/PRBonn/MapMOS). It is as easy to use as 4DMOS and can successfully segment the moving objects for your data, because it uses a local map to identify them:

[image]

It is still not perfect, but at least better than 4DMOS.

curiousityy, 2025-02-25:
@benemer Thank you so much for the quick updates. I will try MapMOS then. The readme said 4DMOS was inspired by mapmos and so I thought this would probably be the best bet. I will set up MapMOS and give it a try. For the sparsity, these were recordings from ouster 64.

Thanks again for all the support and help.

curiousityy, 2025-02-25:
@benemer, you are right. The recordings are originally from os1-64 but were later downsampled to 32. Thanks for that hint. I will test this on the original 64 and with 10 scans ckpt. In parallel, I will also setup MapMOS and give that a try. Thanks for all the help :)

closed 2025-02-25 by benemer