PRBonn/4DMOS #47: No such file or directory error

srmamt00, 2025-03-26:
Hi,

Thanks for the amazing work. I am trying to run this on my custom dataset and I have formatted according to KITTI. For checking, I was using the pre-trained weights provided in your page(10_scans.ckpt). When I give the path for my datasets in the mos4d_pipeline --visualize <weights_path> <data_path>, it is showing error as No such file or directory. I even checked with the KITTI dataset and I am getting the same error. I built 4dmos and minkowski engine using docker. 

I am new to this tool and I am not sure if I am doing errors from my end. And if you can provide me the steps that I need to follow of how should I use this tool for custom dataset, It would be very helpful. 

Thanks in advance for your answer..!!

benemer, 2025-03-26:
Can you post the exact command you are trying to run? Also, what is the folder structure of your dataset?

benemer, 2025-03-26:
Can you please post the output of 

```
ls  /home/path
```

srmamt00, 2025-03-27:
Hi,

Sorry, My bad. There was some conflicts with the docker container. I then built locally. Now I am able to successfully run 4DMOS.

The issue I am facing right now is that still I am able to see the shadows of the dynamic objects even after running the 4DMOS.
I have attached the image fro your reference.

[image]

Steps that I followed:
I used mos4d_pipeline --visualize /home/sriz/Downloads/weights/10_scans.ckpt /home/sriz/Documents/projects/point_cloud_labeler/point_labeler/bin/Datasets/scene_2/velodyne/

After running this, I have extracted the poses results in KITTI format and used that to visualize in point cloud labeler, but still I am able to see that moving vehicles are creating shadows through their trajectory

The contents are as follows:
~/Downloads/weights$ ls
10_scans.ckpt  5_scans_dt_0p2.ckpt    5_scans_single_output.ckpt
2_scans.ckpt   5_scans_dt_0p3.ckpt
5_scans.ckpt   5_scans_no_poses.ckpt

scene_2/velodyne$ ls
000000.bin  000008.bin  000016.bin  000024.bin  000032.bin
000001.bin  000009.bin  000017.bin  000025.bin  000033.bin
000002.bin  000010.bin  000018.bin  000026.bin  000034.bin
000003.bin  000011.bin  000019.bin  000027.bin  000035.bin
000004.bin  000012.bin  000020.bin  000028.bin  000036.bin
000005.bin  000013.bin  000021.bin  000029.bin  000037.bin
000006.bin  000014.bin  000022.bin  000030.bin  000038.bin
000007.bin  000015.bin  000023.bin  000031.bin  000039.bin

Am I missing any steps? What is your suggestion on this?

Thanks in advance for your answer..!!

benemer, 2025-03-27:
This is because you only extract the pose information which does not contain the information about moving objects.

You need to pass the `--save_kitti` option to also save the predicted labels of moving/non-moving. You can then use the [semantic-kitti-api](https://github.com/PRBonn/semantic-kitti-api) for visualizing the scans.

If you need the scans without the moving objects, you can use the `scan_cleaner.py` from [LMNet](https://github.com/PRBonn/LiDAR-MOS) or just modify our `pipeline.py` to dump the filtered scans into any data format you like.

srmamt00, 2025-03-27:
Hi @benemer,

Thank you for the answer. And sorry for the trouble again. 
From --save_kitti, I could save labels of moving /non-moving objects. But in my work, I am focused more on the poses, where in after moving object segmentation, I need to extract the poses and load it into the point cloud labeler for labeling purpose of my custom scene. The output of the pointcloud should not have shadows of moving objects and so labeling will be smooth.

Is there a prescribed way that I could do this with 4DMOS?
Thanks in advance for your answer..!!

benemer, 2025-03-27:
Okay, I see. In that case, you use the poses but still load the raw scans that contain moving objects. Instead, you should use the predictions, filter out the moving points, and save the static point clouds to disk. Then, load the processed point clouds into your labeling tool.

Look at the `scan_cleaner.py` I mentioned above to see how to clean the scans.

srmamt00, 2025-03-27:
Hi,

Thanks for the suggestion. I am currently followed the scan_cleaner.py approach from LMNet. The predictions (.label) files obtained from 4DMOS were loaded. But I am seeing the below error. It seems that 4DMOS is cropping out some points for the predictions.

"Error processing frame 0 in seq 08: boolean index did not match indexed array along dimension 0; dimension is 1105746 but corresponding boolean dimension is 940178
Error processing frame 1 in seq 08: boolean index did not match indexed array along dimension 0; dimension is 1106573 but corresponding boolean dimension is 928399
..... "

Is there a way to rectify this in 4DMOS pipeline?

Thanks in advance..!!

benemer, 2025-03-27:
Yes, by default we clip the scans. Please also pass the path to the KITTI config when running 4DMOS: https://github.com/PRBonn/4DMOS/blob/main/config/kitti.yaml

This will predict moving objects for all points in the scan.

srmamt00, 2025-03-31:
Hi @benemer ,

I followed the steps by loading the KITTI config file while running 4DMOS. 
Steps followed:
mos4d_pipeline --save_kitti --config /home/sriz/Documents/projects/4DMOS/config/kitti.yaml /home/sriz/Documents/projects/4DMOS/weights/10_scans.ckpt /home/sriz/Documents/projects/datasets/test/velodyne/

kitti.yaml file:

data:
  deskew: False

mos:
  max_range_mos: -1.0 # No limit

With this I have generated the predictions (.label) files and I was trying to visualize in Semantic KITTI API and I am getting the following error.
In Semantic KITTI API
 ./visualize_mos.py --sequence 00 --dataset /home/sriz/Documents/Datasets/test_Dataset --predictions /home/sriz/Documents/Datasets/test_Dataset
********************************************************************************
INTERFACE:
Dataset /home/sriz/Documents/Datasets/test_Dataset
Config config/semantic-kitti-mos.yaml
Sequence 00
Predictions /home/sriz/Documents/Datasets/test_Dataset/
ignore_semantics False
do_instances False
ignore_safety False
color_learning_map False
offset 0
********************************************************************************
Opening config file config/semantic-kitti-mos.yaml
Sequence folder exists! Using sequence from /home/sriz/Documents/Datasets/test_Dataset/sequences/00/velodyne
Labels folder exists! Using labels from /home/sriz/Documents/Datasets/test_Dataset/sequences/00/predictions
Using semantics in visualizer
Points shape:  (1119126, 3)
Label shape:  (1105746,)
Traceback (most recent call last):
  File "/home/sriz/Documents/projects/semantic-kitti-api/./visualize_mos.py", line 165, in <module>
    vis = LaserScanVis(scan=scan,
  File "/home/sriz/Documents/projects/semantic-kitti-api/auxiliary/laserscanvis.py", line 31, in __init__
    self.update_scan()
  File "/home/sriz/Documents/projects/semantic-kitti-api/auxiliary/laserscanvis.py", line 137, in update_scan
    self.scan.open_label(self.label_names[self.offset])
  File "/home/ttsadasm6/Documents/projects/semantic-kitti-api/auxiliary/laserscan.py", line 235, in open_label
    self.set_label(label)
  File "/home/sriz/Documents/projects/semantic-kitti-api/auxiliary/laserscan.py", line 251, in set_label
    raise ValueError("Scan and Label don't contain same number of points")
ValueError: Scan and Label don't contain same number of points

Similar error I am facing when I try to use scan_cleaner.py in LMNet. 
I even checked the issue as mentioned in #43 but still I am facing the same issue, that's why I am reaching out again.

Thanks in advance for your answer..!!

benemer, 2025-03-31:
Are you sure you are loading the .label files that you generated before? And that the same data is used?

Because when running 4DMOS, you load the data from `/home/sriz/Documents/projects/datasets/test/velodyne/` and for the SemanticKITTI API, the data is at ` /home/sriz/Documents/Datasets/test_Dataset`.

srmamt00, 2025-04-04:
Hi @benemer,
Thanks. You're right. I messed up in the path specification. I then corrected that and I am able to visualize the masks in kitti api.

The below is the image that I could visualize in kitti api

[image]

Next Steps , I took was to remove the dynamic objects using scan_cleaner.py in LIDAR-MOS. Then I was able to get the clean scans(.bin files) and when I visualize the clean scans in kitti api as well as point cloud labeler, still I am able to visualize the dynamic objects.

The below is the image that I could visualize in kitti api and I could still see the dynamic objects (The vehicle infront of the origin is a dynamic ones)

[image]

The below is the image from point cloud labeler

[image]

The below is the config file from LIDAR-MOS

# Inputs
# the root of raw LiDAR scans
scan_root: 'data'
# the root of mos predictions
mos_pred_root: 'data/predictions_salsanext_residual_1_valid'
# the root of semantic predictions
semantic_pred_root: 'data/predictions_salsanext_sem_valid'

# Outputs
split: valid  # choose from (train, valid, test)
combined_results_root: 'data/predictions_salsanext_residual_1_sem_valid'
clean_scan_root: 'data'

Here I moved the .label files generated from the 4DMOS into mos_pred_root. The clean scans were generated in clean_scan_root.
I used that clean scans to visualize in the kitti api.

These are the steps that I followed. Kindly let me know if I am missing any steps in between, so that is why I am still able to see dynamic objects.

Thanks in advance for your answer..!!

benemer, 2025-04-04:
The point cloud (.bin) you visualized looks like it contains points from multiple scans. Is that correct? For 4DMOS, to properly work, each file must be a single scan.

closed 2025-04-23 by benemer